P3TERX/Actions-OpenWrt #70: SSH 进行配置时，.config 文件已被创建

qianxu2001, 2020-02-08:
在 SSH 上一步的时候会创建一个默认配置的 .config（如果没上传 .figcon 时），这时如果在 SSH 内直接 make menuconfig 就会在原生成的配置上改，导致取消 luci/app 不会自动取消相关依赖，建议在 SSH 脚本那里先运行一次删除 .config，另外可以在教程那里写一下进入网页 SSH 后按一次 Ctrl+C，不然一片黑，输什么命令也没用，我第一次用的时候一脸懵逼

leopardciaw, 2020-02-15:
谢谢分享。如何使用SSH进入哈，取消注释后，修改那两个用户名密码吗，如果是，用户名和密码是github的吗。谢谢！

leopardciaw, 2020-02-15:
> 在 SSH 上一步的时候会创建一个默认配置的 .config（如果没上传 .figcon 时），这时如果在 SSH 内直接 make menuconfig 就会在原生成的配置上改，导致取消 luci/app 不会自动取消相关依赖，建议在 SSH 脚本那里先运行一次删除 .config，另外可以在教程那里写一下进入网页 SSH 后按一次 Ctrl+C，不然一片黑，输什么命令也没用，我第一次用的时候一脸懵逼

谢谢分享。请问下，如何使用SSH进入哈，取消注释后，修改那两个用户名密码吗，如果是，用户名和密码是github的吗。谢谢！

qianxu2001, 2020-02-15:
@leopardciaw 我用的网页 ssh，打开一个链接就可以输命令，你重新 fork 一次仓库，现在已经不用取消注释了，只需要改一个变量，P3 的博客教程里有说明

leopardciaw, 2020-02-16:
> @leopardciaw 我用的网页 ssh，打开一个链接就可以输命令，你重新 fork 一次仓库，现在已经不用取消注释了，只需要改一个变量，P3 的博客教程里有说明

谢谢回复！

P3TERX, 2020-02-17:
感谢建议，你这个情况我之前没有注意过，等有空了研究

choyung, 2020-02-24:
@P3TERX 
SSH_ACTIONS: ture
设置这样然后到了
SSH connection to Actions0s
Loading
一直在Loading…

qianxu2001, 2020-02-24:
@choyung 网络问题，挂代理就好了

choyung, 2020-02-24:
@qianxu2001 我是路由挂的gfwlist全局代理

qianxu2001, 2020-02-24:
@choyung 这个域名并不在 gfwlist 里，你开全局代理就知道了

closed 2020-02-24 by P3TERX

choyung, 2020-02-24:
@qianxu2001 好的，谢谢，我试试。
那开了全局之后，要重新开始吗？还是只是刷新一下页面就好了？

commit 88a21841 (2020-02-24): 'make defconfig' after SSH, resolve #70