P3TERX/Actions-OpenWrt #88: 如何编译lienol大神的固件呢？

yishan2019, 2020-02-25:
如何编译lienol大神的固件呢？

yishan2019, 2020-02-25:
制作个编译lienol固件的模板吧

M1A6I3L, 2020-02-25:
env:
  REPO_URL: https://github.com/coolsnowwolf/lede
  REPO_BRANCH: master

改成

env:
  REPO_URL: https://github.com/Lienol/openwrt
  REPO_BRANCH: 图中的三选一
[image]

yishan2019, 2020-02-26:
“图中三选一”这句没看懂。这句在哪里改？Default branch这句下面好几句呢，加上一句“dev-lean-lede"吗？

yishan2019, 2020-02-26:
“图中三选一”这句没看懂。这句在哪里改？REPO_BRANCH 这句下面好几句呢，加上一句“dev-lean-lede"吗？

yishan2019, 2020-02-26:
是这样吗？
env:
  REPO_URL: https://github.com/Lienol/openwrt
  REPO_BRANCH: dev-9.07
  CONFIG_FILE: .config
  DIY_SH: diy.sh
  FREE_UP_DISK: false
  SSH_ACTIONS: false
  UPLOAD_BIN_DIR: false
  UPLOAD_FIRMWARE: true
  TZ: Asia/Shanghai

yishan2019, 2020-02-26:
或
REPO_BRANCH: dev-19.07
或
REPO_BRANCH: dev-master

是这个意思吗？

P3TERX, 2020-02-26:
是的，多看教程，多实践。比你问一千遍一万遍有用。

closed 2020-02-26 by P3TERX