P3TERX/Actions-OpenWrt #106: 修改默认登录密码和主题应该如何在diy.sh中实现？

goukey, 2020-03-10:
通过您的脚本，我已经成功实现了自动编译。
DIY.SH里的修改默认ip也大致明白，应该就是替换语句对吧？
我还想编译时修改默认密码，比如lean的是默认是password。而他的config里面是
`set network.$1.password='password'`
如果我直接替换password这个单词肯定不行，请问如何来实现呢？
另外我还想默认修改主题，修改哪里呢？
谢谢

goukey, 2020-03-10:
`sed -i 's/'password'/'123456'/g' openwrt/package/base-files/files/bin/config_generate`
这样能否实现？

lyin888, 2020-03-10:
sed -i 's/root::0:0:99999:7:::/root:$1$V4UetPzk$CYXluq4wUazHjmCDBCqXF.:0:0:99999:7:::/g' /etc/shadow

lyin888, 2020-03-10:
$1$V4UetPzk$CYXluq4wUazHjmCDBCqXF.
这些字符就是密码passwd，把这些字符修改成你密码的字符，如何转换，简单的就是把固件修改成你想要的密码，然后复制粘贴

BtQiang, 2020-03-10:
#删除默认密码
sed -i "/CYXluq4wUazHjmCDBCqXF/d" package/lean/default-settings/files/zzz-default-settings

goukey, 2020-03-11:
多谢 修改密码成功了

那么还有两个问题，修改默认主机名和主题在哪个路径文件？

DFOUD, 2020-03-12:
我找到了修改主题的文件，应该是/etc/config/luci，请问我要修改成argon主题改如何配置diy.sh

goukey, 2020-03-12:
在diy.sh里面我试了几次都不行，最后在配置文件里实现的。
添加你需要的主题包估计你会。
下面这行代码可以取消默认主题：
`sed -i '/set luci.main.mediaurlbase=\/luci-static\/bootstrap/d' feeds/luci/themes/luci-theme-bootstrap/root/etc/uci-defaults/30_luci-theme-bootstrap`
你把它添加到clone代码后面运行。

DFOUD, 2020-03-12:
我只加个一个主题，和默认的加起来是两个，取消掉默认主题来达到我想要的主题的效果

lphgor, 2020-03-21:
把默认bootstrap主题改成material：
```sh
sed -i 's/luci-theme-bootstrap/luci-theme-material/g' ./feeds/luci/collections/luci/Makefile
```

mzxwt, 2020-03-28:
P大有没有兴趣再写一个单独编译插件的模版。有时候编译固件就为添加或者升级一个插件，费时费力。

yang12me, 2020-04-19:
> 多谢 修改密码成功了
> 
> 那么还有两个问题，修改默认主机名和主题在哪个路径文件？

请问而你用什么方法修改的？麻烦贴一下

grootsong, 2020-06-26:
请教各位大神，固件里边的版本号怎么自定义呀？虚拟机编译我知道是修改zzz-default-settings文件。但是Action里边，我在替换ip地址下边添加了一行sed命令，sed -i 's/R20.6.18/R202006/g' package/lean/default-settings/files/zzz-default-settings，替换版本号为R202006。一直通不过编译不出来呀？

closed 2020-10-11 by P3TERX