P3TERX/Actions-OpenWrt #123: 能否更换DIY.sh顺序

rilyuuj, 2020-04-07:
感谢作者，这个仓库很有帮助，文档也写的很详细。使用后有一些想法和疑问，望能解答。
代码新手，可能有些问题比较菜鸟，见谅。
1. 能否将workflow 里的自定义配置放到update/install feed前面？因为有时候需要添加其他仓库的package，在修改feed.conf.default，需要再重新feed一下（如果.config中包含新添加的package的话是否需要重新feed还不清楚）调整步骤后，diy.sh不再需要添加feed命令了。
2. 添加个别package时，如果原仓库地址文件在子目录中的话，有什么办法可以只pull子目录中的文件。
举个例子：比如该仓库https://github.com/vernesong/OpenClash，编译文件在luci-app-openclash中，尝试，将echo "luci-app-openclash/*" >> .git/info/sparse-checkout写入sparse中，但是仍旧会将目录拉取下来，目录结构成为package/lean/luci-app-openclash/luci-app-openclash，造成编译时错误。不知道有什么好实现办法可以代替diy.sh中使用mv * ..
3. workflow中env.FIRMWARE变量是那一部分？因为本人需求只需docker使用的.gz文件，不要其他img上传至release，想改写这个actions/upload-artifact
谢谢，还望各位大佬看到能解答下，谢谢。

P3TERX, 2020-04-07:
1. 这样不利于分清是源码本身拉取错误，还是你自己额外加的东西导致的错误。
2. 直接 git clone 即可：`git clone https://github.com/vernesong/OpenClash package/OpenClash`
3. 取消其他不需要的固件格式。

rilyuuj, 2020-04-08:
> 1. 这样不利于分清是源码本身拉取错误，还是你自己额外加的东西导致的错误。
> 2. 直接 git clone 即可：`git clone https://github.com/vernesong/OpenClash package/OpenClash`
> 3. 取消其他不需要的固件格式。

好的，谢谢大佬及时答复，关于3我的意思不是.config里设置需要什么文件，我想请问有什么思路可以写在脚本里可以将自定义的东西打包release，比如只需要packages或者某个image，分别打包release。

garypang13, 2020-04-08:
```
    - name: Organize files
      id: organize
      if: env.UPLOAD_FIRMWARE == 'true' && !cancelled()
      run: |
        mkdir firmware && find openwrt/bin/targets/*/*/* -maxdepth 0 -name "*uefi-gpt*" -or -name "*combined*" -or -name "sha256sums" | xargs -i mv -f {} ./firmware/
        cd firmware
        echo "::set-env name=FIRMWARE::$PWD"
        echo "::set-output name=status::success"
```
@rilyuuj 我的方法

sky2016cn, 2020-04-13:
谢谢作者的辛苦付出，云编译让我们小白轻松了很多，请教作者指出我是否有错误，云编译过程是修改了REPO_URL: https://github.com/coolsnowwolf/lede这个地址，改为我fork过来的地址，在fork过来的feeds.conf.default里添加了两行src-git lienol https://github.com/Lienol/openwrt-package
src-git lisaac https://github.com/lisaac/luci-app-diskman
，这样也能把这两个插件编译入固件，请问这样方式对吗？添加的两行命令对吗？

rilyuuj, 2020-04-14:
> ```
>     - name: Organize files
>       id: organize
>       if: env.UPLOAD_FIRMWARE == 'true' && !cancelled()
>       run: |
>         mkdir firmware && find openwrt/bin/targets/*/*/* -maxdepth 0 -name "*uefi-gpt*" -or -name "*combined*" -or -name "sha256sums" | xargs -i mv -f {} ./firmware/
>         cd firmware
>         echo "::set-env name=FIRMWARE::$PWD"
>         echo "::set-output name=status::success"
> ```
> 
> @rilyuuj 我的方法
> 
> 上面的例子 先把要打包的找出来放到firmware目录里,然后cd到了firmware目录 然后把当前路径$PWD赋值给了env.FIRMWARE

感谢大佬，最近不知道为什么开启ssh后，一直无法在页面展开查看内容，所以获取不到真实的工作目录。等能登录进去看一次环境情况就明白怎么做了。
关于第二个问题，我后来在diy.sh 按照这个方式去添加软件，但是只有第一个软件添加进去了，不知道是不是写法有什么问题？
```
# OpenClash for Luci
git clone https://github.com/vernesong/OpenClash package/OpenClash
./scripts/feeds update luci-app-openclash
./scripts/feeds install luci-app-openclash
# Kcptun client for Luci
git clone https://github.com/kuoruan/luci-app-kcptun.git package/luci-app-kcptun
./scripts/feeds update luci-app-kcptun
./scripts/feeds install luci-app-kcptun
```

rilyuuj, 2020-04-14:
> 谢谢作者的辛苦付出，云编译让我们小白轻松了很多，请教作者指出我是否有错误，云编译过程是修改了REPO_URL: [...]

是的，可以的。config里面还要选上你需要添加的软件。

sky2016cn, 2020-04-14:
谢谢您的回复，大佬们时时更新的问题我注意到了，每次编译完，全都删除以前的for，所以每次编译都是重新fork一下，也包括你的，还好不是经常编译，谢谢！

P3TERX, 2020-04-24:
现在增加了一个 DIY 脚本，在更新 feeds 之前执行。

closed 2020-04-24 by P3TERX

ocean521, 2020-06-21:
> 1. 这样不利于分清是源码本身拉取错误，还是你自己额外加的东西导致的错误。
> 2. 直接 git clone 即可：`git clone https://github.com/vernesong/OpenClash package/OpenClash`
> 3. 取消其他不需要的固件格式。

大佬，现在OPENCLASH 存在分支，也可以这么写么？