P3TERX/Actions-OpenWrt #125: 奶牛快传好像url拼接有问题什么的还是下载解压有问题，系统是ubuntu-latest

yinzhefuyun, 2020-04-08:
% Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     9  100     9    0     0     72      0 --:--:-- --:--:-- --:--:--    71
100     9  100     9    0     0     72      0 --:--:-- --:--:-- --:--:--    71

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
##[error]Process completed with exit code 2.

ClayMoreBoy, 2020-04-08:
同样出现这个问题 @P3TERX

P3TERX, 2020-04-08:
[image]

closed 2020-04-08 by P3TERX