P3TERX/Actions-OpenWrt #133: 提一个小建议关于显示sha256的小优化

yinzhefuyun, 2020-04-13:
在上传网盘的过程中先cat一下sha256sums文件，把sha256的内容打印出来，这样就不用下载一个固件刷机包和一个文件校验文件，在web更新的时候还可以对比下sha256是否有出入
    - name: Upload firmware to cowtransfer
      if: steps.organize.outputs.status == 'success' && env.UPLOAD_COWTRANSFER == 'true' && !cancelled()
      run: |
        curl -fsSL git.io/file-transfer | sh
        ./transfer cow --block 2621440 -s -p 99 -t 3 --hash --no-progress ${FIRMWARE} 2>&1 | tee cowtransfer.log
        echo "::warning file=cowtransfer.com::$(cat cowtransfer.log | grep https)"
        **cat ${FIRMWARE}/sha256sums**

cmsax, 2020-04-27:
如果这类对于编译出来的固件需求比较多的话，建议写成类似于 diy 一样的代码，有种生命周期的感觉。具体可以参考代码中加入 diy-part1 和 diy-part2 的写法，在固件编译之后加一个 diy-part3，欢迎贡献 PR。

cmsax, 2020-04-27:
你可以在第 118 行加入以下命令：

```shell
chmod -x ../diy_part3.sh
../diy_part3.sh
```

然后在根目录添加 `diy_part3.sh`，其中文件内容：

```shell
cd openwrt
shasum -a 256 $FIRMWARE > shasum.txt
```

closed 2020-10-11 by P3TERX