P3TERX/Actions-OpenWrt #136: 大佬，好像不小心把奶牛快传的脚本换成了WeTransfer

yinzhefuyun, 2020-04-14:
https://github.com/P3TERX/Actions-OpenWrt/commit/5290d4c1d7448e30f0b7fdd622e66c57b7dc7c8a

    - name: Upload firmware to cowtransfer
      if: steps.organize.outputs.status == 'success' && env.UPLOAD_COWTRANSFER == 'true' && !cancelled()
      run: |
        **curl -fsSL git.io/file-transfer | sh**脚本居然是同一个
        ./transfer cow --block 2621440 -s -p 99 -t 3 --hash --no-progress ${FIRMWARE} 2>&1 | tee cowtransfer.log
        echo "::warning file=cowtransfer.com::$(cat cowtransfer.log | grep https)"

    - name: Upload firmware to WeTransfer
      if: steps.organize.outputs.status == 'success' && env.UPLOAD_WETRANSFER == 'true' && !cancelled()
      run: |
        **curl -fsSL git.io/file-transfer | sh**脚本居然是同一个
        ./transfer wet -s -p 16 --no-progress ${FIRMWARE} 2>&1 | tee wetransfer.log
        echo "::warning file=wetransfer.com::$(cat wetransfer.log | grep https)"

hxchtiewei, 2020-04-15:
大佬调用的是集合多个API的大文件传输工具，所以是同一个。
只是我用奶牛会丢文件，除了第一个大文件我能下载，其他都提示“{"error":"Document not found"}”
https://c-t.work/s/a45faf1bf97949

yinzhefuyun, 2020-04-15:
> 大佬调用的是集合多个API的大文件传输工具，所以是同一个。
> 只是我用奶牛会丢文件，除了第一个大文件我能下载，其他都提示“{"error":"Document not found"}”
> https://c-t.work/s/a45faf1bf97949

好像是GitHub自己下载不到的错误curl: (6) Could not resolve host: git.io 敢相信吗

closed 2020-04-15 by yinzhefuyun