P3TERX/Actions-OpenWrt #148: download package 报错

CrazyMouse, 2020-04-27:
流程执行到make defconfig的时候报一下错误：
Run cd openwrt
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b

#
# configuration written to .config
#
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..cbba577f6dcbffcd5a2933baad48459ec3f5681b
time: target/linux/prereq#0.08#0.02#0.10

请问这个会影响最终编译结果么？

closed 2020-04-29 by CrazyMouse

abc2251259, 2020-09-19:
同样的问题，请问你解决了吗？

CrazyMouse, 2020-09-22:
> 同样的问题，请问你解决了吗？

没问题，忽略就好