P3TERX/Actions-OpenWrt #153: 请问是否只有UPLOAD_FIRMWARE为true才能上传到奶牛或者WETRANSFER？

pirately, 2020-05-06:
请问是否只有UPLOAD_FIRMWARE为true才能上传到奶牛或者WETRANSFER？
即，如果单独设置UPLOAD_COWTRANSFE或者UPLOAD_WETRANSFER为true，但是UPLOAD_FIRMWARE: false，实际上奶牛或者WETRANSFER不能上传？
能否设置成不上传到GitHub本身的actions，只上传到奶牛等？

garypang13, 2020-05-07:
> 能否设置成不上传到GitHub本身的actions，只上传到奶牛等？

可以的

pirately, 2020-05-07:
> > 能否设置成不上传到GitHub本身的actions，只上传到奶牛等？
> 
> 可以的

请问，怎么实现？因为我在GitHub私有仓库设置UPLOAD_FIRMWARE: false，后续的UPLOAD_COWTRANSFE或者UPLOAD_WETRANSFER就算设置成true，都没有上传。

garypang13, 2020-05-07:
把第118行 if: env.UPLOAD_FIRMWARE == 'true' && !cancelled() 删掉即可

pirately, 2020-05-07:
> 把第118行 if: env.UPLOAD_FIRMWARE == 'true' && !cancelled() 删掉即可

感谢，我试试

Hey-Leo, 2020-05-15:
> 把第118行 if: env.UPLOAD_FIRMWARE == 'true' && !cancelled() 删掉即可

似乎不删除，把`&&`换成`||`更好一点

closed 2020-05-23 by P3TERX

Hey-Leo, 2020-06-05:
### 在build-openwrt.yml中
> Organize files （line127）

-  删掉 `if: env.UPLOAD_FIRMWARE == 'true' && !cancelled()` ，

 
>Upload firmware directory （line 136）

-  `if: steps.organize.outputs.status == 'success' && !cancelled()` 改为 `if: env.UPLOAD_FIRMWARE == 'true' && !cancelled()`