P3TERX/Actions-OpenWrt #159: 出错 Check failure on line 1 in .github  @github-actions github-actions / build  .github#L1 No space left on device

y162287761, 2020-05-21:
请问出现
 Check failure on line 1 in .github

@github-actions
github-actions
/ build

.github#L1
No space left on device

 Check failure on line 1 in .github

@github-actions
github-actions
/ build

.github#L1
Process completed with exit code 2.

 Check warning on line 1 in .github

@github-actions
github-actions
/ build

.github#L1
Failed to remove 'http.https://github.com/.extraheader' from the git config

请问出现这些错误怎么解决

aizhouyi, 2020-05-21:
我也是这样的错误，七八次了。
no space left on device

yangxiangguo, 2020-05-22:
我也是  如何解决

CORE2021, 2020-05-22:
同样！空间不足了？

Itxiaoniao, 2020-05-22:
Check failure on line 1 in .github

@github-actions
github-actions
/ build

.github#L1
No space left on device

aizhouyi, 2020-05-22:
我在lean那里看到同样的issue了！！！源码问题？

P3TERX, 2020-05-23:
最近 Actions 做了一些改动，把工作目录挂载到了 14G 临时分区，所以导致了空间不足。

不过这个改动导致了一些问题（与空间不足无关），今天已经回滚了，所以目前已经**暂时**恢复到了可正常编译 OpenWrt 的状态。

actions/virtual-environments#709
coolsnowwolf/lede#4654

closed 2020-05-23 by P3TERX

aizhouyi, 2020-05-23:
编译成功了，今天。感谢回答原因。