P3TERX/Actions-OpenWrt #164: 建议把lean源码中的ssr plus调出来

xiechangan123, 2020-06-06:
如题，还要自己手动调出来，默认被lean隐藏了。希望大佬把调出来代码写进你的脚本里面，免除我等的时间和脑力成本，万分感谢
贴出来自己搜的相关文章：https://ssr.tools/1420

Hey-Leo, 2020-06-06:
取消掉 `diy-part1.sh` `第_12_行的注释#不就行了`

Hey-Leo, 2020-06-06:
取消掉 `diy-part1.sh` 第12行的开头注释`#`不就行了

Hey-Leo, 2020-06-06:
还有，这不是被隐藏，是去掉了这个功能

Hey-Leo, 2020-06-06:
还有，这不是被隐藏，lede没有这什么花里胡哨的东西

hxchtiewei, 2020-06-10:
diy-part1.sh里已有，你要自己去除#

xiechangan123, 2020-06-11:
> diy-part1.sh里已有，你要自己去除#

嗯，成功调出来了。那我想问这个版本与https://github.com/fw876/helloworld上的哪个版本新？或者说去除#后直接就是git clone fw876的，所以版本一直最新？主要是不会看代码写的软链接

DingXiangYuanZhiXing, 2020-06-14:
@xiechangan123 Actions-OpenWrt 是直接uncomment L大`feeds.conf.default`里面的这一行`#src-git helloworld https://github.com/fw876/helloworld`。 所以一直是最新的。

Dimitri2020007, 2020-08-31:
修改L大lede的feeds.conf.default中#src-git helloworld https://github.com/fw876/helloworld代码前的注释符号就可以调出ssr-plus

closed 2020-10-11 by P3TERX