P3TERX/Actions-OpenWrt #165: 编译报错，请指教

steveny5997, 2020-06-07:
查了下日志，是这个内容：

make[2]: *** [package/install] Error 255
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
package/Makefile:65: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:108: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

感谢各位大神！

be-engineer, 2020-07-11:
me too,不知道是什么原因造成的，郁闷呀

closed 2020-10-11 by P3TERX

Derek-G, 2020-10-26:
这个问题解决了么就close了？ 我也遇到相同的问题了，google 一大圈也没看到办法