P3TERX/Actions-OpenWrt #177: 开启luci-app-ssr-plus是不是这么弄？另外还想把openclash也装上去，应该怎么弄？

jif739519, 2020-06-20:
diy-part1.sh
sed -i '$a src-git helloworld https://github.com/fw876/helloworld' feeds.conf.default

开启luci-app-ssr-plus，就是这么弄？

liuran001, 2020-06-21:
OpenClash  
添加到diy-part2.sh  
`git clone -b master https://github.com/vernesong/OpenClash package/openclash`
.config  
`CONFIG_PACKAGE_luci-app-openclash=y`  
ssr同理  
当然也可以在diy-part1.sh中修改，但是较繁琐  
你可以参考下[我的配置](https://github.com/liuran001/Openwrt_MI-R3G_firmware)

jif739519, 2020-06-23:
> OpenClash
> 添加到diy-part2.sh
> `git clone -b master https://github.com/vernesong/OpenClash package/openclash`
> .config
> `CONFIG_PACKAGE_luci-app-openclash=y`
> ssr同理
> 当然也可以在diy-part1.sh中修改，但是较繁琐
> 你可以参考下[我的配置](https://github.com/liuran001/Openwrt_MI-R3G_firmware)

感谢大佬，另外有个问题，有airprint隔空打印吗？

jif739519, 2020-06-23:
另外：
# Modify default IP
#sed -i 's/192.168.1.1/192.168.50.5/g' package/base-files/files/bin/config_generate

比如要修改默认ip为192.168.123.222，网关192.168.123.1，掩码255.255.255.0，应该怎么搞？

liuran001, 2020-06-23:
> 另外：
> #Modify default IP
> #sed -i 's/192.168.1.1/192.168.50.5/g' package/base-files/files/bin/config_generate
> 
> 比如要修改默认ip为192.168.123.222，网关192.168.123.1，掩码255.255.255.0，应该怎么修改上面的例句？

我建议你直接用files大法（因为我也不太会改config）   
直接把修改好的配置导出然后写入固件

liuran001, 2020-06-23:
> > OpenClash
> > 添加到diy-part2.sh
> > `git clone -b master https://github.com/vernesong/OpenClash package/openclash`
> > .config
> > `CONFIG_PACKAGE_luci-app-openclash=y`
> > ssr同理
> > 当然也可以在diy-part1.sh中修改，但是较繁琐
> > 你可以参考下[我的配置](https://github.com/liuran001/Openwrt_MI-R3G_firmware)
> 
> 感谢大佬，另外有个问题，有airprint隔空打印吗？

airprint是啥emmmm  
还是usb打印服务器?

jif739519, 2020-06-24:
> airprint是啥emmmm
> 还是usb打印服务器?

airprint隔空打印，用于iOSS设备直接连接，google系也有类似的，进行打印

liuran001, 2020-06-24:
> airprint隔空打印，用于iOSS设备直接连接，google系也有类似的，进行打印

看了下，驱动就要几百MB，我路由是搞不得😂😂😂

cs-star, 2020-08-17:
最简单同添加.config一样，点击Create new file（创建新文件）文件名填写为feeds.conf.default ，把生成的feeds.conf.default 。

closed 2020-10-11 by P3TERX