P3TERX/Actions-OpenWrt #204: 最新版本使用--depth 1参数克隆的仓库编译会发生错误

blackknifes, 2020-08-30:
如题，如果加上--depth 1 会导致仓库仅克隆最新版本的代码，而不包含历史代码，这回导致以下脚本执行错误，从而导致在固件重启时，所有配置被重置：
./scripts/getver.sh

Dimitri2020007, 2020-08-31:
#你需要修改/workflow里的脚本
#在checkout下添加内容(name是我改的)
#steps:
    - name: 校验本地文件
      uses: actions/checkout@main
      with:
        ref: master（可加可不加，没有影响）
        fetch-depth: 0（这个选项可以控制源码复制，不加默认为1，git自动克隆最近1次的修改源码，0就是所有的。）

#改完了记得把git clone后面的--depth 1删除
这样就可以复制所有openwrt源了

blackknifes, 2020-08-31:
我在自己fork的库下已经删除了，但是我认为这个修改应该合并到官方仓库中，因为openwrt 19.X 以上的库都会出现这样的问题，添加--depth 也并没有节省太多的时间。

Dimitri2020007, 2020-08-31:
我编译的是L大的lede，openwrt我还没编译

closed 2020-09-07 by blackknifes

xiasonming, 2020-09-11:
> [，]，\，\，\，\，\，\，\，\，\，\，\，\，\，\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\\

> 如题，如果加上--depth 1 会导致仓库仅克隆最新版本的代码，而不包含历史代码，这回导致以下脚本执行错误，从而导致在固件重启时，所有配置被重置：
> ./scripts/getver.sh

感谢你提的问题，让我找到了编译出来的官方固件刷机无法正常启动的问题