P3TERX/Actions-OpenWrt #212: 在线编译出错

Dimitri2020007, 2020-09-18:
[image]

脚本出错信息

2020-09-17T22:07:46.8939448Z  make[2] package/install
2020-09-17T22:08:56.8536857Z make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2020-09-17T22:08:56.8538264Z /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
2020-09-17T22:08:56.8542047Z make: *** [world] Error 1
2020-09-17T22:09:05.0338689Z make[1]: Entering directory '/workdir/openwrt'
2020-09-17T22:09:06.8523425Z make[2]: Entering directory '/workdir/openwrt'
2020-09-17T22:09:08.6316960Z make[3]: Entering directory '/workdir/openwrt/target/linux'
2020-09-17T22:09:08.6557710Z make[4]: Entering directory '/workdir/openwrt/target/linux/x86'
2020-09-17T22:09:08.7197262Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/localversion
2020-09-17T22:09:08.7221561Z /workdir/openwrt/scripts/kconfig.pl  + + /workdir/openwrt/target/linux/generic/config-5.4 /workdir/openwrt/target/linux/x86/config-5.4 /workdir/openwrt/target/linux/x86/64/config-5.4 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target
2020-09-17T22:09:08.7731029Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target
2020-09-17T22:09:08.7822251Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target
2020-09-17T22:09:08.7839506Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target
2020-09-17T22:09:08.7855657Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target
2020-09-17T22:09:08.7880370Z /workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 5.4 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.override
2020-09-17T22:09:08.8823537Z /workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.target /dev/null /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.override > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set
2020-09-17T22:09:08.9383056Z mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.old
2020-09-17T22:09:08.9410220Z grep -v INITRAMFS /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.old > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set
2020-09-17T22:09:08.9447992Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set
2020-09-17T22:09:08.9467394Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set
2020-09-17T22:09:08.9483513Z rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
2020-09-17T22:09:08.9507166Z cmp -s /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.prev || { cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config; cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.prev; }
2020-09-17T22:09:08.9551694Z export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/user_headers ] || make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61 KCFLAGS="-ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Sep 16 12:04:43 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.61 CC="x86_64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/user_headers headers_install
2020-09-17T22:09:08.9565113Z grep '=[ym]' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.config.set | LC_ALL=C sort | mkhash md5 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.vermagic
2020-09-17T22:09:08.9611805Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/.configured
2020-09-17T22:09:08.9636843Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/vmlinux /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61/System.map
2020-09-17T22:09:08.9662077Z make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61 KCFLAGS="-ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl" HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Sep 16 12:04:43 2020" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.61 CC="x86_64-openwrt-linux-musl-gcc" modules
2020-09-17T22:09:08.9687028Z make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.61'

892947707, 2020-09-18:
我这里提示
An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'

Dimitri2020007, 2020-09-18:
你的make menuconfig生成的.config上传到脚本里能编译出固件吗？

892947707, 2020-09-18:
> 你的make menuconfig生成的.config上传到脚本里能编译出固件吗？虚拟机编译能通过，脚本咋就不行了？

就是不行我才来看看的，启动十几秒就报错了，昨天都没问题的

elonj, 2020-09-18:
> 我这里提示
> An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'

遇到同样问题

weweyes, 2020-09-18:
> > 我这里提示
> > An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'
> 
> 遇到同样问题

同样问题，大家如何处理？

elonj, 2020-09-19:
> > > 我这里提示
> 
> > > An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'
> 
> > 
> 
> > 遇到同样问题
> 
> 
> 
> 同样问题，大家如何处理？

不需要用到SSH的话，可以先把SSH相关的部分注释掉。或者等修复

892947707, 2020-09-19:
> > > > 我这里提示
> > 
> > 
> > > > An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'
> > 
> > 
> > > 
> > 
> > 
> > > 遇到同样问题
> > 
> > 
> > 同样问题，大家如何处理？
> 
> 不需要用到SSH的话，可以先把SSH相关的部分注释掉。或者等修复

我没有使用SSH

db-one, 2020-09-19:
看上去好像是P3TERX创建了一个什么脚本，大家所有的自动编译都是通过这个来运行具体编译的，现在这个脚本出问题了，所以全跪了吧

db-one, 2020-09-19:
https://api.github.com/repos/P3TERX/debugger-action/tarball/master

有人了解这个api脚本怎么建立的吗？可以自己搞一个？

elonj, 2020-09-19:
> > > > > 我这里提示
> > > 
> > > 
> > > > > An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'
> > > 
> > > 
> > > > 
> > > 
> > > 
> > > > 遇到同样问题
> > > 
> > > 
> > > 同样问题，大家如何处理？
> > 
> > 
> > 不需要用到SSH的话，可以先把SSH相关的部分注释掉。或者等修复
> 
> 我没有使用SSH
哦哦，那就不清楚是什么原因了。我把SSH相关的部分注释掉之后就没有报错了
[image]
[image]

badcrazy, 2020-09-19:
> > > > > > 我这里提示
> > > > 
> > > > 
> > > > > > An action could not be found at the URI 'https://api.github.com/repos/P3TERX/debugger-action/tarball/master'
> > > > 
> > > > 
> > > > > 
> > > > 
> > > > 
> > > > > 遇到同样问题
> > > > 
> > > > 
> > > > 同样问题，大家如何处理？
> > > 
> > > 
> > > 不需要用到SSH的话，可以先把SSH相关的部分注释掉。或者等修复
> > 
> > 
> > 我没有使用SSH
> 
> 哦哦，那就不清楚是什么原因了。我把SSH相关的部分注释掉之后就没有报错了
> [image]
> [image]

搞定，谢谢

Dimitri2020007, 2020-09-19:
将ssh那块改成false，无需注释。就在开头。最好有make menuconfig生成的.config文件。在源码根目录下创建同名文件，将.config里内容全粘贴过去。最好有config.buildinfo文件。可以直接复制到源码根目录的.config中。

closed 2020-09-28 by Dimitri2020007