P3TERX/Actions-OpenWrt #214: 编译出错了，前几天都没有问题的

weweyes, 2020-09-19:
各位大佬怎么解决这个问题？
[image]

badcrazy, 2020-09-19:
都一样

gongz5481, 2020-09-19:
都一样。看来最近风控很严。只能选择本地编译了

Dimitri2020007, 2020-09-19:
是和ssh有关吗？
env:
  REPO_URL: https://github.com/coolsnowwolf/lede
  REPO_BRANCH: master
  FEEDS_CONF: feeds.conf.default
  CONFIG_FILE: .config
  DIY_P1_SH: diy-part1.sh
  DIY_P2_SH: diy-part2.sh
  SSH_ACTIONS: false
  UPLOAD_BIN_DIR: false
  UPLOAD_FIRMWARE: true
  UPLOAD_COWTRANSFER: false
  UPLOAD_WETRANSFER: true
  TZ: Asia/Shanghai

把ssh_actions改成false再试试

weweyes, 2020-09-20:
> ssh_actions改成false

ssh_actions改成false还是不行

gongz5481, 2020-09-20:
是大神的api的问题。需要github重新设置。所以和ssh无关。目前只有等待修复

Dimitri2020007, 2020-09-20:
等api修复了再试试吧，实在有编译需求的去搭全局和一个虚拟机执行本地编译

qxzg, 2020-09-21:
把ssh的那几行注释掉就可以了，可以参考我的：https://github.com/qxzg/My-OpenWRT/commit/2c69e0db6720425be2a69239270098ae8b84b96e

gongz5481, 2020-09-22:
> 把ssh的那几行注释掉就可以了，可以参考我的：[qxzg/My-OpenWRT@2c69e0d](https://github.com/qxzg/My-OpenWRT/commit/2c69e0db6720425be2a69239270098ae8b84b96e)

这个只是权宜之计.对于需要使用到ssh的小白，还是要等大神重新设置api才可以

gongz5481, 2020-09-23:
action@master改成action@main 可以成功ssh编译

closed 2020-09-25 by weweyes