P3TERX/Actions-OpenWrt #235: 固件编译图形化界面

nextplusapp, 2020-10-12:
Dear P3TERX,

首先非常感谢大神的贡献，极大地简化了固件编译方面环境搭建成本。为了进一步降低固件编译的学习成本和技术门槛，特别为本项目开发了一个图形化界面，地址如下：

https://openwrt-build.github.io/

ps: 目前还处于测试阶段，可能不是很完善，欢迎大家测试，提出您宝贵的意见和建议，谢谢！

Dimitri2020007, 2020-10-14:
可以考虑添加以下功能来完善你的图形化编译界面

1.添加.config之类的选项可以更好的定制固件的软件包或者kmod还有其他的东西

2.添加固件git源的选项（用户可以从官方openwrt、lean的lede、lienol的openwrt的git源自由选择进行编译）

3.添加软件包和组件冲突，功能重复的提醒，避免小白走弯路。（比较费功夫且不太现实，可以忽略的）

4.添加第3方git源的功能（可以自由的定制固件源码中没有的主题、或者其他的需求是固件源码中没有的。）

暂时只有这么多

nextplusapp, 2020-10-15:
> 可以考虑添加以下功能来完善你的图形化编译界面
> 
> 1.添加.config之类的选项可以更好的定制固件的软件包或者kmod还有其他的东西
> 
> 2.添加固件git源的选项（用户可以从官方openwrt、lean的lede、lienol的openwrt的git源自由选择进行编译）
> 
> 3.添加软件包和组件冲突，功能重复的提醒，避免小白走弯路。（比较费功夫且不太现实，可以忽略的）
> 
> 4.添加第3方git源的功能（可以自由的定制固件源码中没有的主题、或者其他的需求是固件源码中没有的。）
> 
> 暂时只有这么多

1. 上传自定义.config文件，已经在开发计划中了
2. 这个是个很好的想法，后续会考虑支持
3. 因为menuconfig的限制，没有办法脚本化，所在做Web UI的确是有一些困难，不过一直在做技术上的探索
4. 这个也已经在开发计划中了，如果你有好的第三方包的推荐，麻烦贴上来，我们方便做一个UI界面。

非常感谢您的宝贵意见！

Dimitri2020007, 2020-10-16:
第3方主题源有这些：

1.https://github.com/apollo-ng/luci-theme-darkmatter.git
2.https://github.com/openwrt-develop/luci-theme-atmaterial.git
3.https://github.com/xiaoqingfengATGH/luci-theme-infinityfreedom.git
4.https://github.com/tano-systems/luci-theme-tano.git

1.luci-darkmatter主题（黑色、部分仿碳纤维的界面）
2.luci-atmaterial主题（强烈推荐、界面跟koolshare lede很像的）
3.luci-infinityfreedom主题（强烈推荐、界面精致很漂亮的主题）
4.luci-tano主题（白色、界面也是很漂亮的）

第3方软件包推荐：

1.https://github.com/Lienol/openwrt-OpenAppFilter.git
2.https://github.com/xiaorouji/openwrt-package.git
3.https://github.com/fw876/helloworld.git

1.是lienol制作的基于openwrt的应用程序过滤模块
2.是xiaorouji制作的passwall插件（包含部分openwrt软件包，与lienol openwrt中passwall不一样的地方是他完善了passwall的naiveproxy的bug。）
3.是fw876制作的ssr+插件（2.是速度快，3.是稳定，若与lean的lede一起使用则要加上一条指令才能打开ssr+应用，这个可以谷歌找到。openwrt ssr plus）

可能有用户会有不一样的体验，在2和3上，有些用户觉得2比3稳定而且快，还有一部分用户觉得3比2快而且稳定。
这个应该视地区、运营商和线路差异的问题。

目前推荐的第3方软件包暂时就这么多

而且该脚本目前只匹配了官方openwrt和lean的openwrt。
要匹配lienol的openwrt需要多下点功夫。（我用这个脚本编译lienol的openwrt的时候卡在编译环节了。总是1小时左右就出错了。报错是在工具链编译完成以后出的。）

nextplusapp, 2020-10-17:
收到，@Dimitri2020007 非常感谢！
HelloWorld默认已经包含

moqingzhu1, 2021-01-08:
大神增加一个星际宝盒吧

nextplusapp, 2021-01-19:
> 大神增加一个星际宝盒吧

请提供一下项目地址，谢谢！

closed 2021-01-30 by nextplusapp