P3TERX/Actions-OpenWrt #242: action脚本编译lienol openwrt报错

Dimitri2020007, 2020-10-23:
报错位置是工具链编译完成，开始编译软件时候出现的
在实体机中编译没有问题

错误日志如下：

2020-10-20T11:53:23.5083622Z Console input/output is redirected. Run 'make oldconfig' to update configuration.
2020-10-20T11:53:23.5084100Z 
2020-10-20T11:53:23.5084767Z scripts/kconfig/Makefile:38: recipe for target 'silentoldconfig' failed
2020-10-20T11:53:23.5085418Z make[7]: *** [silentoldconfig] Error 1
2020-10-20T11:53:23.5086003Z make[6]: *** [silentoldconfig] Error 2
2020-10-20T11:53:23.5086711Z Makefile:527: recipe for target 'silentoldconfig' failed
2020-10-20T11:53:23.5113393Z 
2020-10-20T11:53:23.5121943Z The present kernel configuration has modules disabled.
2020-10-20T11:53:23.5130725Z Type 'make config' and enable loadable module support.
2020-10-20T11:53:23.5138418Z Then build a kernel with module support enabled.
2020-10-20T11:53:23.5146875Z 
2020-10-20T11:53:23.5154687Z make[5]: *** [modules] Error 1
2020-10-20T11:53:23.5156327Z Makefile:1314: recipe for target 'modules' failed
2020-10-20T11:53:23.5159250Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.200'
2020-10-20T11:53:23.5161809Z make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.200/.modules] Error 2
2020-10-20T11:53:23.5163893Z Makefile:30: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.200/.modules' failed
2020-10-20T11:53:23.5166336Z make[3]: *** [compile] Error 2
2020-10-20T11:53:23.5167528Z make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
2020-10-20T11:53:23.5169643Z make[2]: *** [target/linux/compile] Error 2
2020-10-20T11:53:23.5170784Z Makefile:13: recipe for target 'compile' failed
2020-10-20T11:53:23.5171887Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2020-10-20T11:53:23.5172516Z time: target/linux/compile#20.08#6.86#28.97
2020-10-20T11:53:23.5173336Z target/Makefile:23: recipe for target 'target/linux/compile' failed
2020-10-20T11:53:23.5174352Z make[2]: Leaving directory '/workdir/openwrt'
2020-10-20T11:53:23.5189472Z target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
2020-10-20T11:53:23.5210876Z make[1]: Leaving directory '/workdir/openwrt'
2020-10-20T11:53:23.5213860Z make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2020-10-20T11:53:23.5225405Z /workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-10-20T11:53:23.5230661Z make: *** [world] Error 2
2020-10-20T11:53:23.5234083Z ##[error]Process completed with exit code 2.

bigboy323, 2020-10-28:
错误同上，脚本不能用了吗？

Dimitri2020007, 2020-10-28:
脚本可以成功编译lean openwrt和官方openwrt，就是不知道为啥lienol的openwrt不成功，工具链一编译完就不行了。

Swiftfrog, 2020-11-07:
我也是同样的问题，不知道有没有解决方法？

CyberTNT, 2020-11-09:
编译lienol的openwrt除了更换github源地址，上传.config文件之外还要做以下几件事：
①对比发现编译前要用到的软件有些不一样（即apt-get install所安装的那些），要参照换成lineol的readme中所写的那些
②lienol仓库的branch默认不是master，要改成已有的分支
③对比发现lienol的编译步骤中多了一步clean，参照lineol的readme增加这一步骤

CyberTNT, 2020-11-09:
> 脚本可以成功编译lean openwrt和官方openwrt，就是不知道为啥lienol的openwrt不成功，工具链一编译完就不行了。

我自己改了一下workflow文件成功编译了，你可以拿我仓库里的workflow文件拉到你自己的仓库编译试试

Dimitri2020007, 2020-11-10:
我修改了一下代码，软件包、分支、clean都改过了。

与软件包无关，不管用的是lean readme下的软件包还是lienol readme下的软件包，一样都会出错。

但在虚拟机下编译成功。

经过仔细检查，出错的位置是那样一致，（已经测试数回action）总是在那个位置出错。

2020-11-10T06:12:01.0265229Z   Histogram triggers (HIST_TRIGGERS) [N/y/?] (NEW) aborted!（第一回）

2020-11-10T05:42:32.3393871Z   Histogram triggers (HIST_TRIGGERS) [N/y/?] (NEW) aborted!（第二回）

2020-11-10T05:10:00.0549828Z   Histogram triggers (HIST_TRIGGERS) [N/y/?] (NEW) aborted!（第三回）

三回的错误那么一致，是不是哪里还要修改？

2020-11-10T05:10:00.0549828Z   Histogram triggers (HIST_TRIGGERS) [N/y/?] (NEW) aborted!
2020-11-10T05:10:00.0550124Z 
2020-11-10T05:10:00.0550818Z Console input/output is redirected. Run 'make oldconfig' to update configuration.
2020-11-10T05:10:00.0554366Z 
2020-11-10T05:10:00.0554741Z make[7]: *** [silentoldconfig] Error 1
2020-11-10T05:10:00.0555614Z scripts/kconfig/Makefile:38: recipe for target 'silentoldconfig' failed
2020-11-10T05:10:00.0558485Z make[6]: *** [silentoldconfig] Error 2
2020-11-10T05:10:00.0559727Z Makefile:527: recipe for target 'silentoldconfig' failed
2020-11-10T05:10:00.0578168Z 
2020-11-10T05:10:00.0587645Z The present kernel configuration has modules disabled.
2020-11-10T05:10:00.0597696Z Type 'make config' and enable loadable module support.
2020-11-10T05:10:00.0608125Z Then build a kernel with module support enabled.
2020-11-10T05:10:00.0618375Z 
2020-11-10T05:10:00.0628610Z make[5]: *** [modules] Error 1
2020-11-10T05:10:00.0629884Z Makefile:1314: recipe for target 'modules' failed
2020-11-10T05:10:00.0631721Z make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.202/.modules] Error 2
2020-11-10T05:10:00.0632724Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.202'
2020-11-10T05:10:00.0634103Z make[3]: *** [compile] Error 2
2020-11-10T05:10:00.0635022Z Makefile:30: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-4.14.202/.modules' failed
2020-11-10T05:10:00.0636234Z make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
2020-11-10T05:10:00.0637159Z Makefile:13: recipe for target 'compile' failed
2020-11-10T05:10:00.0638025Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2020-11-10T05:10:00.0638652Z time: target/linux/compile#1.36#0.46#1.79
2020-11-10T05:10:00.0657460Z make[2]: *** [target/linux/compile] Error 2
2020-11-10T05:10:00.0658392Z target/Makefile:23: recipe for target 'target/linux/compile' failed
2020-11-10T05:10:00.0659713Z make[2]: Leaving directory '/workdir/openwrt'
2020-11-10T05:10:00.0685243Z make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2020-11-10T05:10:00.0686283Z target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
2020-11-10T05:10:00.0687137Z make: *** [world] Error 2

CyberTNT, 2020-11-11:
> 我修改了一下代码，软件包、分支、clean都改过了。
>
> 与软件包无关，不管用的是lean readme下的软件包还是lienol readme下的软件包，一样都会出错。
>
> 但在虚拟机下编译成功。
>
> 经过仔细检查，出错的位置是那样一致，（已经测试数回action）总是在那个位置出错。
>
> 2020-11-10T06:12:01.0265229Z Histogram triggers [...]

看了一下报错它应该是让你运行make oldconfig来加入你修改过的内核特性吧，在make阶段加一步make oldconfig试试

CyberTNT, 2020-11-11:
还有小问一下：一开始你说还要diffconfig什么的，你不是有虚拟机吗，生成一个改好的.config传上去不就行了，为啥弄那么麻烦

CyberTNT, 2020-11-11:
实在不行就用这个https://github.com/281677160/build-openwrt，这个脚本作者做了专属lineol的action脚本，改的比较好

Dimitri2020007, 2020-11-11:
我用的就是改好的.config，如果连.config都编译不了我还要diffconfig干嘛啊

Dimitri2020007, 2020-12-11:
分支rebase19.07-5.4是可以在线编译的

closed 2020-12-11 by Dimitri2020007