P3TERX/Actions-OpenWrt #243: 编译出来的固件很大

Jsonylq, 2020-10-26:
通过虚拟机调整好.config并编译，编译后的固件在20M左右(包含helloworld)，然后将配置放到云编译中，diy-part1.sh放开了helloworld，最后编译完的固件快40M，求问各位大佬这个是什么原因导致的呢，有什么好的方法和建议吗？

Ljzkirito, 2020-10-26:
我看了你的[Jsonylq/Opwrt_YKL1](https://github.com/Jsonylq/Opwrt_YKL1/actions/runs/327442429)，
Artifacts下的是zip压缩包，包括
openwrt-ramips-mt7620-youku_yk1.manifest
openwrt-ramips-mt7620-youku_yk1-initramfs-kernel.bin                18.4MB
openwrt-ramips-mt7620-youku_yk1-squashfs-sysupgrade.bin        19MB
config.buildinfo
feeds.buildinfo
version.buildinfo
sha256sums

建议你下载下来看看，然后关闭此Issue。
可以开启`UPLOAD_RELEASE`，直接在release下载squashfs-sysupgrade.bin单个文件

Jsonylq, 2020-10-26:
> 我[看了](https://github.com/Jsonylq/Opwrt_YKL1/actions/runs/327442429)你的[Jsonylq / Opwrt_YKL1](https://github.com/Jsonylq/Opwrt_YKL1/actions/runs/327442429)，
> 工件下是的压缩包，包括
> openwrt-ramips-mt7620-youku_yk1.manifest
> openwrt-ramips-mt7620-youku_yk1-initramfs-kernel.bin **_18.4MB_**
> openwrt-ramips_mty761- squashfs-sysupgrade.bin _**19MB**_
> config.buildinfo
> feeds.buildinfo
> 版本
> .buildinfo sha256sums
> 
> 建议您下载下来看看，然后关闭此问题。
> 可以打开`UPLOAD_RELEASE`，直接在发行版下载squashfs-sysupgrade.bin单独文件

抱歉没仔细看，以为编译出来的是单个文件，实际是两个文件。非常感谢！

closed 2020-10-26 by Jsonylq