P3TERX/Actions-OpenWrt #247: 请教大佬这编译是哪出错了？

asdfvv, 2020-10-30:
/workdir/openwrt/include/image.mk:356: recipe for target '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/root.ext4' failed
3419
make[5]: Leaving directory '/workdir/openwrt/target/linux/armvirt/image'
3420
Makefile:20: recipe for target 'install' failed
3421
make[4]: Leaving directory '/workdir/openwrt/target/linux/armvirt'
3422
Makefile:13: recipe for target 'install' failed
3423
make[3]: Leaving directory '/workdir/openwrt/target/linux'
3424
time: target/linux/install#233.37#22.86#263.97
3425
target/Makefile:23: recipe for target 'target/linux/install' failed
3426
make[2]: *** [target/linux/install] Error 2
3427
make[2]: Leaving directory '/workdir/openwrt'
3428
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install' failed
3429
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_install] Error 2
3430
make[1]: Leaving directory '/workdir/openwrt'
3431
make: *** [world] Error 2
3432
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
3433
Error: Process completed with exit code 2.

yinyouzhu, 2020-11-26:
同遇到这个问题

michzhan, 2021-01-16:
我使用openwrt官方的，如果分支直接指向master，也遇到这个错误。
后来分支指向v19.07.5就没有碰到这个问题，我估计master默认最新分支可能配置那里有点问题。
希望有大佬能解决。
目前建议还是使用稳定的release 来编译

以上是我的经验，仅供参考。如果你使用的不是官方的源，可能得再看看了

yinyouzhu, 2021-01-22:
请问如何重新指向其他分支呢

closed 2021-07-08 by P3TERX