P3TERX/Actions-OpenWrt #280: 大佬，编译openwrt卡到这里不动了

aabbcc582, 2021-01-27:
Run P3TERX/ssh2actions@v1.0.0
[INFO] Setting up tmate ...
[INFO] You may need to enter a password to authorize.

[INFO] Check the architecture ...
[INFO] Check the version of tmate ...
[INFO] Download tmate ...
[INFO] tmate download successful !
[INFO] Installation tmate ...
[INFO] tmate successful installation !
[INFO] Running tmate...[](url)
[image]
[image]

bikizz, 2021-01-27:
同样的问题，昨天编译可以跳链接，今天试了一直卡

winona2021, 2021-02-21:
我也碰到了一样的问题，一个新的repo，编译成功过一次之后，之后的每次编译都卡在了“ssh connect to actions”

superabbite1, 2021-04-06:
我也是卡ssh

closed 2021-07-08 by P3TERX