P3TERX/Actions-OpenWrt #288: 编译提示proxy.golang.com.cn无法下载

iseeyou, 2021-02-14:
Finding targets
go: downloading github.com/golang/protobuf v1.4.3
go: downloading github.com/v2fly/v2ray-core v4.34.0+incompatible

Building targets
go: downloading github.com/golang/protobuf v1.4.3
go: downloading github.com/v2fly/v2ray-core v4.34.0+incompatible
../../main.go:16:2: github.com/golang/protobuf@v1.4.3: Get "https://proxy.golang.com.cn/github.com/golang/protobuf/@v/v1.4.3.zip": stopped after 10 redirects
../../main.go:18:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:20:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:37:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:21:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:22:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:23:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../errors.generated.go:3:8: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:38:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:25:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:40:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:26:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:27:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:29:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:30:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:32:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:33:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:34:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects
../../main.go:35:2: github.com/v2fly/v2ray-core@v4.34.0+incompatible: Get "https://proxy.golang.com.cn/github.com/v2fly/v2ray-core/@v/v4.34.0+incompatible.zip": stopped after 10 redirects

Makefile:74: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-1.8.0/.built' failed
make[3]: Leaving directory '/workdir/openwrt/package/lean/v2ray-plugin'
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-1.8.0/.built] Error 1
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
time: package/lean/v2ray-plugin/compile#0.60#0.17#6.01
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

ChrisLiu0343, 2021-02-14:
同样问题…

iseeyou, 2021-02-14:
> 同样问题…

国内电脑编译也是这问题

ChrisLiu0343, 2021-02-14:
> > 同样问题…
> 
> 国内电脑编译也是这问题

我PC挂梯可以正常编译完成但是云编译这两天怎么也过不去，以前没有这个问题。

piaoyizy, 2021-02-15:
make clean

closed 2021-02-15 by iseeyou