P3TERX/Actions-OpenWrt #309: 关于build-openwrt.yml 中出现安装 "qq" 指令的疑惑？

iHub-2020, 2021-05-15:
为什么在build-openwrt.yml中的第 48 - 55 行之间出现了安装qq的指令：

        sudo rm -rf /etc/apt/sources.list.d/* /usr/share/dotnet /usr/local/lib/android /opt/ghc
        sudo -E apt-get -qq update
        sudo -E apt-get -qq install $(curl -fsSL git.io/depends-ubuntu-1804)
        sudo -E apt-get -qq autoremove --purge
        sudo -E apt-get -qq clean
        sudo timedatectl set-timezone "$TZ"
        sudo mkdir -p /workdir
        sudo chown $USER:$GROUPS /workdir

想请问大佬这里为什么需要安装qq这种让人发毛的软件呢？还请帮忙解惑，谢谢啦！

qxzg, 2021-05-15:
-qq是减少输出的意思
https://serverfault.com/questions/644180/what-does-qq-argument-for-apt-get-mean

iHub-2020, 2021-05-15:
@qxzg 知道了，非常感谢:)

Dimitri2020007, 2021-07-02:
apt -qq选项等于-y和-q（静默安装）

closed 2021-07-08 by P3TERX