P3TERX/Actions-OpenWrt #328: 编译L大OpenWRT固件，要安装ubuntu-20.04

feizaibufei, 2021-07-01:
我是按照Actions-OpenWrt修改了下，可以编译固件老是出错。麻烦大佬解决一下
[image]
[image]
[image]
[image]

 sudo -E apt-get install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch python3 python2.7 unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler g++-multilib antlr3 gperf wget curl swig rsync

zwillhill, 2021-07-02:
在自己仓库新建文件，文件名任意，内容从`build-essential`开始到依赖结束，
将文件raw链接（链接格式如`https://raw.githubusercontent.com/***`）在[git.io](https://git.io/)网页换成短链，
生成的短链替换build-openwrt.yml中的短链`git.io/depends-ubuntu-1804`

feizaibufei, 2021-07-02:
老是到了Compile the firmware这步提示
This step has been truncated due to its large size. Download the full logs from the menu once the workflow run has completed.

zwillhill, 2021-07-02:
> 老是到了Compile the firmware这步提示
> This step has been truncated due to its large size. Download the full logs from the menu once the workflow run has completed.

又看了一眼你的截图，发现你apt-get后面的-qq没了
源码中的命令示例如下
 `sudo -E apt-get -qq update`

其中的-E 和 -qq 我也还没查到具体含义，但我选择保留，编译是没有问题的（上午刚编译过，改过depends和ubuntu版本）。
当然在ubuntu实机中测试没有-E和-qq也是能正常编译的

建议装个beyondcompare之类的，本地对比下自己的代码和源代码，很容易发现区别。

closed 2021-07-08 by P3TERX